ArduPilot/MissionPlanner #1428: "Unexpected error" trying to upload mission

night-ghost, 2016-12-14:
I'm at home, getting ready to fly, of course no GPS. Clock on 'Write WP' and got

[image]

meee1, 2016-12-14:
please set your home location. right hand side under the write wp button.

night-ghost, 2016-12-14:
What a speed of closing!

Why I need to set FAKE home location when I try  just to write WPs? I do not know where I'm going to fly, so the return is to be on the point of Arm, not to there where I write WP.

night-ghost, 2016-12-14:
moreover, it would be better to make a more clear diagnosis than "unexpected error"

meee1, 2016-12-14:
because home is sent as part of the wp upload. always has been.

ill look at making the error nicer

night-ghost, 2016-12-14:
fresh controller, city apartment - MP may not have the coordinates! In this case, it is more logical to send NAN, NAN as coordinates -  the controller to knew that there is no Home point

meee1, 2016-12-14:
MP requires you to set a home point for many of the display features/distance calcs. so no, home is REQUIRED.

night-ghost, 2016-12-14:
But this REQUIRED value can be a NAN, yeah? MP is rather a tool for configuring and not for flight control, while Tover is more suitable for the direct support of the flight.

So, it is illogical when the Configuration Tool for some reason requires coordinates. In the best case there is no coords, at worst, these coordinates are wrong, and the copter will remembers that point of "home" and fly away.

rmackay9, 2016-12-14:
Just to be clear though, aside from whatever argument about functionality, I don't think Copter will fly away when doing the RTL.  the vehicle's home is normally set on arming.  It is possible to set it from the ground station as well, but I'm unsure whether that is set when someone does/doesn't click on the home fields in MP.

night-ghost, 2016-12-14:
I wish to hear it! So, you agree that the requirement to specify the home point to upload mission is excessive because it will be replaced on Arming?

meee1, 2016-12-14:
@night-ghost you can set your home as 0,0,0 if you want. that is your choice, and MP will accept that.

meee1, 2016-12-14:
added message here
https://github.com/ArduPilot/MissionPlanner/commit/c836f7f2c8b1ba0cd21f533b1083de135500b994

closed 2016-12-14 by meee1