ArduPilot/MissionPlanner #1432: Joystick button assignement

Julien-768, 2016-12-15:
#### Issue details
Joystick button assignement are not saved with Auto-detect

Pitch and Throttle are inverted and are not corresponding to the mapping made in the RCMAP parameters.

#### Version
Mission planner 1.3.41.8

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
Rover - Navio 2

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2016-12-28:
when you are setting up the joystick ,are you connected?
the settings might be being saved under the wrong airframe type.

Julien-768, 2017-01-17:
Yes it's while I'm connected

meee1, 2017-01-17:
is it still happening in the latest release?