ArduPilot/MissionPlanner #1437: Copter: LOG_WHEN_DISARMED Checkbox doesn't work with AC3.4

R-Lefebvre, 2016-12-22:
The LOG_DISARMED checkbox in the Standard Parameters page does not toggle the LOG_DISARMED parameter.  Instead is probably toggles the old logging bitmask bit.

meee1, 2016-12-28:
ive updated the base param meta data
https://github.com/ArduPilot/MissionPlanner/commit/ec2c4b25c686f52d92843e2bd1e22ee85abfae33
however if you had done an update, this should have updated itself.
either way problem should be resolved with newest version

closed 2016-12-28 by meee1

rmackay9, 2016-12-29:
Great, thanks for this.

R-Lefebvre, 2016-12-30:
Sorry, what do you mean "done an update"?  I think my MP install was updated.

meee1, 2016-12-30:
if you had done an update since installing from either zip or msi, it should have resolved itself