ArduPilot/MissionPlanner #1438: Logfile variables explaination

BurnPlane, 2016-12-27:
Hi everyone, where can I find the specific explaination of logfile variables for fixed-wing? Fixed-wing logfile variables are different from copter',like AHR2,ARSP,ATT,BARO,ect. thanks.
[image]

magicrub, 2016-12-27:
HowTo questions are best asked in the forum: discuss.ardupilot.org but since I know they are in the wiki, but old, I also suggest starting an issue on the wiki in GitHub too.

BurnPlane, 2016-12-27:
thanks,magicrub, i'm a newbee for github, don't know how to start an issue, but still thanks.

magicrub, 2016-12-27:
This is an issue. You just made one in mission planner.

rmackay9, 2016-12-28:
I might close this one.  It's not really a mission planner thing, but rather a Plane wiki thing..

closed 2016-12-28 by rmackay9

blowschoolup, 2019-07-29:
have you found the answer,i have the same question now