ArduPilot/MissionPlanner #1440: GCS Link Quality indicator incorrect with MavProxy redundant connection

custeve, 2016-12-28:
Using MavProxy, I have two links to the same Pixhawk (APM plane 3.7.1): 
Link 0: USB-->SOCAT (on OpenWRT)-->UDP port 14506 over WiFi Mesh -->MavProxy
Link 1: 3dr 915Mhz radio --> MavProxy

Mavproxy command is: 
mavproxy.py --master=udpin:0.0.0.0:14506 --master=/dev/ttyUSB0 --out=udpbcast:192.168.10.255:14550

Mission Planner (1.3.43) connects using UDP to 14550, and that works fine. These connections work very well individually, the link quality indicator in MP reads accurately. However, when I use it as above, in a redundant configuration, the link quality indicator reads 3% (though it's clear by observing the status tab that the mavlink stream is coming through just fine). It also fails over and re-establishes just fine when either connection is lost. 

My guess is that something in the mavlink sequence is different between the merged USB and 3DR connections, causing MP to misjudge the link quality. There is also a possibility the issue is with MavProxy.

meee1, 2016-12-28:
what happens when you connect both to mission planner at the same time?
to do this just connect to the first normally, then right click the top menu bar and pick connection options, and connect the seconds link

meee1, 2016-12-28:
the underlying issue is that mavproxy is multiplexing the 2 streams together. anything received of that stream will show very poor/random sequence id's.
MP can handle the streams as 2 separate streams, and will report correctly.
I don't really see an answer to this issue as it stands at the moment.

magicrub, 2016-12-28:
This question has come up a few times. I wonder if the is a way to detect
two intermixed streams?

custeve, 2016-12-28:
Thanks, Michael and Tom.

I was afraid that would be the answer. However, I was under the impression that redundant connections to the vehicle is not a new thing. In the current configuration, MP seems to be working just fine, only the link quality reading seems effected. I can try the dual links directly from MP, but I think we know what to expect. Unfortunately, that’s not a feasible solution for our application. Perhaps a better failover method in MavProxy would appropriate, or writing our own proxy script that accomplishes the same thing. Or other suggestions?

Thanks much,

-          Steve

magicrub, 2016-12-29:
How about using different sysid's?