ArduPilot/MissionPlanner #1442: Update Frame Type screen to support FRAME_CLASS, FRAME_TYPE

rmackay9, 2016-12-31:
For AC3.5 we are introducing a FRAME_CLASS parameter and renaming FRAME to FRAME_TYPE so it would be good to update the Install Setup >> Mandatory Hardware >> Frame Type screen.

Possible values from FRAME_CLASS are 0:Undefined, 1:Quad, 2:Hexa, 3:Octa, 4:OctaQuad, 5:Y6, 6:Heli, 7:Tri, 8:SingleCopter, 9:CoaxCopter
Possible values for FRAME_TYPE are 0:Plus, 1:X, 2:V, 3:H, 4:V-Tail, 5:A-Tail, 10:Y6B

[image]

billbonney, 2016-12-31:
What is Undefined? is that a real selectable value? What values are valid combinations? 

Y6B was the most 'stable' at the time. Is Y6-H a valid config for example?

I think the UI will need bit more thought to be useful. It may just be set of valid combinations and not just a 1-to-1 mapping of parameter settings.

rmackay9, 2016-12-31:
FRAME_CLASS will be "Undefined" when multicopter users first load the multicopter firmware on a new board.  So they will need to pick a valid alternative.
It's true that there are some combinations that are not valid.  Like a Quad+Y6B.  There are also some combinations that we could/should implement easily like Octa+H to reduce the number of invalid choices.

currently the only way to know the valid options is to look at the big switch statement here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp#L378

rmackay9, 2017-05-11:
this is done so closing.

closed 2017-05-11 by rmackay9