ArduPilot/MissionPlanner #1446: 3DR PixHawk Mini

Croninburg, 2017-01-06:
Hi 

I have recently picked up one of 3DR's PixHawk Mini and I am trying to configure it using mission planner. 

I am struggling to get mission planner to recognize the PixHawk Mini and load the firmware onto it. 

Is there a know issue or has support for 3DR's PixHawk not been added yet? 

Any help would be greatly appreciated 

Regards
Chris

proficnc, 2017-01-06:
It is supported, it's just a normal Pixhawk in the eyes of Mission planner.

Best you contact 3DR support.

meee1, 2017-01-06:
@Croninburg how did you install mission planner? what os? and are the drivers installed?

rmackay9, 2017-01-06:
If you're flying Copter then from a flight code point of view it's not supported with the latest official release AC3.4.4.  Support is in "master" so I think plane and rover may support it already and Copter will support it with the next release, AC3.5 which is going into beta testing this month.

Croninburg, 2017-01-12:
Hi Guys

Thanks for all the replies. 

I am using Windows 10, 
I installed the latest version on MP (1.43) using the .msi file. 
All relevant drivers have been installed. 
I am trying to set up the PixHawk mini for a quad-copter

I have now attempted using AC3.5, but I get a preArm error of "check firmware / Frame Class" I believe that this is due not being able to run through the wizard process and selecting the correct frame layout. I did try search for a Frame class parameter in under the config tab but i couldn't seem to find one. 

When I do run the wizard after loading the AC3.5 firmware it seems to load firmware AC3.4 which doesn't work, as it fails to get past the accelerator calibration stage. 

Thanks for the help 

Regards
Chris

proasian8118, 2017-02-08:
i have same problem with you Croninburg. i have install frimware px4 and it's working with Qgroucontrol, but it's not working in mission planner FW, i try to use mission planner but i don't know how now... randy we need your help !!!

proficnc, 2017-02-08:
you must load Ardupilot, not PX4

proficnc, 2017-02-08:
@Croninburg 
Under initial setup, mandatory settings. there is a frame section. select it here.

rmackay9, 2017-02-08:
Ah, very nice.  I hadn't actually seen the new selector in MP.  But there it is, at least in the beta version of mission planner.  Here's a picture.
[image]

If you can't find that for some reason, set the FRAME_CLASS parameter to "1" for Copter.

Support for AC3.5 is here: http://discuss.ardupilot.org/c/arducopter/copter-3-5
Wiki: http://ardupilot.org/copter/index.html

proasian8118, 2017-02-08:
i know, i mean it's working with Qgroundcontrol fw but it's not work with adrupilot frimware. i'm using for airplane. i was install AP 3.7.1 and backup to 3.6 3.4 it's not work same. the board keeping red light, it's turn to yellow light just when i uprade to 3.7.1 AP.

proasian8118, 2017-02-08:
here is the answer from 3dr support this morning.
Thank you for contacting 3DR Technical Support.
As per the quick start guide mentions, please use QGroundControl as the pixhawk mini was made for PX4 firmware and does not support APM firmware.

rmackay9, 2017-02-08:
@proasian8118,
ah, you're trying to use plane are you?  It's just a matter of getting a new enough firmware.  The "latest" and "beta" versions certainly support it but not the current "official".  You're trying to load "3.8.0beta2".  There's a little "Beta Firmwares" link on the bottom right of the MP's install firmware screen.

proasian8118, 2017-02-10:
@rmackay9 
Dear randy, thanks you. i just update and it's work well, but i can't reversed channel in MP, i only can reversed on my radio controller. just now can't do anything...

DrummyFloyd, 2017-07-11:
hi , 
i've some trouble with my pixhawk mini, 
i tried to update it with the last software( 3.4.6) and put the bdr_type to 5 
and i've still the a issues on it 

**Check BRD_TYPE: No MPU9250 found**

someone had the same issue? 
thank  for all !

magicrub, 2017-07-12:
@DrummyFloyd please post help questions in the forum http://discuss.ardupilot.org instead of the developer TODO list and bug tracking system.