ArduPilot/MissionPlanner #1449: Plane: take advantage of new waypoints in flight planning

magicrub, 2017-01-07:
Thanks to @EShamaev 's great work (https://github.com/ArduPilot/ardupilot/pull/5403) implementing my issue (https://github.com/ArduPilot/ardupilot/issues/3753) we can have improved mission planning for photo missions with less waypoints. When planning a grid, we can now overshoot a waypoint and then circle back to the next one with just zig zap points, no additional over-shoot waypoints.

Here's a great image of that from the PR:
[image]

WickedShell, 2017-01-07:
You know that MP handled that overshoot for you already on survey grids without extra waypoints and is arguably better as the GCS is more accurately visualizing the planned mission.

magicrub, 2017-01-07:
Oh?

EShamaev, 2017-01-07:
There is a feature on generating photo grid to have overshoot, but it clutters the flight plan with extra points. When having large photo mission this doubles the size of mission plan. Also it is much harder to change it manually after generation because of such extra points.
I think the proposed method to use standard mavlink message payload is much cleaner and correct way to have this behavior.

WickedShell, 2017-01-07:
Must be a regression as it used to not generate extra waypoints, and I
routinely do it without any extra waypoints.

I'm interested why you think that editing the per field is easier then the
entire mission since you would have to do it on a per item basis, which
isn't easier then just regenerating the grid (if anything its more error
prone). After all you wouldn't apply the overshoot to an entire mission
uploaded it, just part of it, otherwise planning something like a landing
would be come impossible.

EShamaev, 2017-01-07:
I believe that generation of waypoints for photo utilizing new method much better describes the mission itself.
The overfly distance when generating mission with AutoWP Survey Grid can use the new method, instead of shifting the points away from polygon.

Also these 2 parameters are standard in mavlink and can be used not only for the purpose of photo missions. Photo missions by AutoWP Survey Grid is only one specific case where it can be used.

WickedShell, 2017-01-07:
I'm not opposing the issue/that it went in, I do like supporting MAVLink
features. I'm just noting that I can't see how it has actually improved
survey grids at all, and that I can't think of a single use case where
simply shifting the planned point would have accomplished the same thing
for no more effort, with better user visualization/prediction of where the
aircraft would be.

EShamaev, 2017-01-07:
I think it's a very subjective opinion and I agree that in photo survey grids the same can be achieved with shift and it is already implemented in MP.
But I personally don't like this, as the mission points no longer represent the real survey area. Especially after the flight, when doing review, or going over the same flight plan in a month or 2, proposed new method provides better visualization of the area that is really being surveyed.
As for me, I treat mission points as a general flight plan, not the exact position of the plane. For example, new autolanding algorithm that is widely discussed now should have some internal decisions made and will no longer represent the real position or behavior of the plane during landing. There are other examples of such things either already implemented or discussed to be implemented.

iskess, 2017-07-09:
It would be very helpful for visualization to have a marker placed on the flight plan path to indicate these WP parameters. For "Acc radius" is would show a small marker the set distance before the WP, and for "Pass by" it would show on the set distance after the WP. This should display on the Flight Data map as well as the Flight Plan map, so the operator can know when the plane is expected to initiate its turn. There could even be an indication on the HUD next to the WP Dist that shows the offset of when the turn will start. + means "Pass by" distance, and - means "Acc radius" distance. For example, 500m to WP2 with a 'Pass by" of 15 would look like: 2>500+15

Naterater, 2018-12-14:
After using mission planner for years, I happened to stumble on this tonight. This would make missions MUCH simpler and save roughly 1/2 of the waypoints when using this overshoot feature. It looks like it's already there, but there's no documentation anywhere, and I finally discovered the column headings because https://github.com/ArduPilot/MissionPlanner/issues/2019 has been fixed.

[image]