ArduPilot/MissionPlanner #1450: Inject GPS with TCP Client has input string format problem

messerjani, 2017-01-08:
Dear Developers,

I've also jumped into RTK-GPS navigation for my surveying quadcopter with Emlid Reach kit. During following Emlid tutorial step-by-step for APM integration (https://docs.emlid.com/reach/apm-integration/) I got an error message in Mission Planner when it's asked for define TCP Client hostname or IP.

Image with screenshot about error is attached. Unfortunately, some parts of the text are in Hungarian (default Windows language on my laptop) but I think main problem is in the second row, it says: Format of input character string is not valid.
[image]

I got always the same error message and never saw incoming RTK-corrections. I'm using latest version MP 1.3.43.2 with installed beta updates and ArduCopter 3.4.4 firmware. Reach base and rover are working properly, checking them in ReachView app.

Thank your for your help.

meee1, 2017-01-08:
which screen are you inputting numbers into?

the issue will be , vs . to fix it I just want to check which box the error is originating from

meee1, 2017-01-08:
my guess is its the numbers in the ublox m8p setup. duration or accuracy.

messerjani, 2017-01-09:
Oh, yes the , vs . was the problem in SurveyIn Acc value. I forgot that in Hungarian Windows comma is the default separator, sorry about that.
Thank you for your help.

closed 2017-01-10 by messerjani