ArduPilot/MissionPlanner #1451: many proximity views appearing

rmackay9, 2017-01-10:
With a proximity sensor attached (which send DISTANCE_SENSOR messages), MP seems to be creating many proximity views.  Sometimes it's just two but sometimes it's about 6.  Maybe there's some kind of race condition between when the viewer fully pops up and when then sensor-distance messages arrive?

meee1, 2017-01-21:
I have a feeling it would have been because MP was parsing old tlogs. I cant 100% confirm that though

rhvonlehe, 2017-04-13:
A related bug is that the BackColor of the forms varies.  In other words, I might get 2 proximity windows that have a black background and 1 that has a white background.  It looks like the Proximity form is not setting BackColor, which means it is inherited from its parent.  In this particular case, my MP must be instantiating 2 forms from one parent and the 3rd form from a different parent.