ArduPilot/MissionPlanner #1454: Flight Plan's Spline view is straight for some tracks

rmackay9, 2017-01-12:
In the current beta release I think something funny may have happened to the spline view.  If you look at the track from 2 to 3 in the image below, it appears straight for some reason.  Also the track from 4 to 5 is not straight but instead has a slight bend half way along it's length.
[image]

meee1, 2017-01-12:
i cant explain it, but something is wrong with the Grad %. I just tested here, and it looks ok
[image]

meee1, 2017-01-12:
somehow home alt is factoring in...

meee1, 2017-01-12:
the cause is because you have the altitude mode set to absolute. as relative it returns to normal. I will check what is going on
ie absolute home of 900 to absolute 25m

meee1, 2017-01-12:
after looking further, the result is correct.
the 3d velocity includes a very large z axis change, so the projected path is less noticeable on the x/y axis.

magicrub, 2017-01-12:
I've been bit by this in the past too. Since most people fly using
relative, would it be possible to post a warning msg once per boot if
waypoints are found in absolute mode?

meee1, 2017-01-12:
on upload MP check if the user meant absolute or not for this exact case

meee1, 2017-01-12:
@magicrub do you know how long ago it was you had issues
MP has checked since the end of 2015
https://github.com/ArduPilot/MissionPlanner/blame/master/GCSViews/FlightPlanner.cs#L2032

meee1, 2017-01-12:
make that start of 2013
https://github.com/ArduPilot/MissionPlanner/blob/8b2c5105c820e542d3fdc9668991763d78306a5d/GCSViews/FlightPlanner.cs#L1266

magicrub, 2017-01-12:
Eh? Was that tag for @rmackay9?

meee1, 2017-01-21:
closing, as is displaying correctly.

closed 2017-01-21 by meee1