ArduPilot/MissionPlanner #1455: JDrones jD-RF868Plus / RF Design 868+ Modem not working with SIK Radio configurator

bobotus, 2017-01-12:
**Tested on:**
-Latest stable build
-Latest beta build 1.3.43.3 build 1.1.6220.1108

**Issue:**
Settings not populated correctly and grayed out. 

**Mission Planner output:**
[image]

**Modem output on RealTerm:**
`+++`
`ATI`
`ATI2`
`ATI3`
`ATI4`
`ATI5`
`ATI6`
`ATI7`
[image]

**Unit works with RFDTools-V1.6 configurator:**
http://files.rfdesign.com.au/tools/
[image]

**Links:**
http://store.rfdesign.com.au/rfd-868-modem/
http://store.jdrones.com/jD_RD868Plus_Telemetry_Bundle_p/rf868set02.htm
http://ardupilot.org/copter/docs/common-3dr-radio-advanced-configuration-and-technical-information.html

meee1, 2017-01-20:
could you please upload the missionplanner.log file?
just run mp, try get the settings > close MP
and send me the file
c:\programdata\missionplanner\missionplanner.log

meee1, 2017-01-21:
the issue is that the ATI5 commands no longer return a value starting with S
ie S1:param:value
ive updated the code, but don't know why the convention changed in the first place.

meee1, 2017-01-21:
change was here https://github.com/ArduPilot/MissionPlanner/commit/631d5e750f297fb89609224a3d40c1672d45b26c

closed 2017-01-21 by meee1

MikkoSaarisalo, 2017-02-06:
Hi There,  I just did hit into similar problem.  

MissionPlanner 1.3.44 build 1.1.6240.11550

Now I am able to load the parameters and see setting populated into the left side of the Sik Radio setup in MissionPlanner.  Trying to modify them does result into an error message box saying "Set Command error"

MissioinPlanner.log attached.
[MissionPlanner.log.txt](https://github.com/ArduPilot/MissionPlanner/files/754990/MissionPlanner.log.txt)

meee1, 2017-02-06:
please update mp ti 1.3.44

MikkoSaarisalo, 2017-02-06:
I am using this one:  1.3.44 build 1.1.6240.11550

meee1, 2017-02-06:
your error is the radio erroring for some reason
 (C:\Users\michael\Source\Repos\MissionPlanner\Radio\Sikradio.cs:1122) [Base Thread]
2017-02-06 17:17:56,350  INFO MissionPlanner.Sikradio - Doing Command AT3=2
 (C:\Users\michael\Source\Repos\MissionPlanner\Radio\Sikradio.cs:1084) [Base Thread]
2017-02-06 17:17:56,365  INFO MissionPlanner.Sikradio - ERROR
 (C:\Users\michael\Source\Repos\MissionPlanner\Radio\Sikradio.cs:1122) [Base Thread]

I don't know why it would be doing that

what firmware do you have on the radio?

MikkoSaarisalo, 2017-02-06:
I'll double check tomorrow and edit it here once done (approx 10 hours from now).   If I recall right, it is RFDSiK V1.13 rfd900p.ihx.

meee1, 2017-02-06:
please try the latest beta MP

MikkoSaarisalo, 2017-02-07:
On version 1.3.44.2  build 1.1.6247.13882

Local modification of net ID is now succesful.  

Modifying both, local and remote does not report an error, but leads to loosing connectivity to remote.  It seems that only the local one actually does get updated.  After playing a with it for a while, I am not able to connect to the remote any more by just changing the Net ID back and forth.  
[MissionPlanner.log.localChangeSuccess.txt](https://github.com/ArduPilot/MissionPlanner/files/756886/MissionPlanner.log.localChangeSuccess.txt)

[MissionPlanner.log.localAndRemoteChangeFishy.txt](https://github.com/ArduPilot/MissionPlanner/files/756887/MissionPlanner.log.localAndRemoteChangeFishy.txt)

meee1, 2017-02-07:
according to those logs you never had a valid remote connection in the first place.
ie rti5 never returned anything valid at all

MikkoSaarisalo, 2017-02-07:
That is correct.  In the second log the remote radio is already in a state where I just could not establish a radio connection there any more.  The way to recover was to connect it through USB to the RFD´s own configuration application twice.  At the first load of the radio configuration the RFD´s application crashes and on the second load the application reads everything gracefully and also writing the settings does work.  Somehow assigning a new Net ID remotely managed to get the radio(s)? to some broken state,

This seems to be repeatable pattern.  

After having updated "recovered" the radio with RFD tool, it does respond gracefully also to MissionPlanner SiK radio configuration.  Including reading the remote configuration to the MissionPlanner display.  I did not try any more to make remote configurations as I need to get my system out to work.    Will be able to test with a spare unit later though if necessary.