ArduPilot/MissionPlanner #1460: Swap position of 'Onboard Mag Calibration" and "Missio nPlanner Mag Calibration"

rmackay9, 2017-01-21:
We want people to use the Onboard Mag Calibration over the mission planner version if possible.

Swapping the position might help push them towards the onboard calibration.  Perhaps we should try and make it more obvious that this produces better results.  By the way, we now have a capabilities bit that can be used to ensure that the vehicle supports to the onboard method.  I suppose you could hide the "mission planner" version if that bit is on.  or perhaps highlight the onboard one?

Unfortunately we only added the capabilities bit in AC3.4.4 but there are a few versions before that that support the onboard calibration but didn't have the capabilities bit.

OXINARF, 2017-01-21:
Related to #1422

meee1, 2017-01-21:
done under #1459

closed 2017-01-21 by meee1

sn0west, 2017-04-23:
Just updated my MP and the onboard calibration is gone. Why?

I got arduplane 3.7.1 and the latest MP.