ArduPilot/MissionPlanner #1461: Fail to connect to Pixhawk using COM port (Using mono on Ubuntu)

tatsuy, 2017-01-21:
Hello,
I use Mission Planner using mono on Ubuntu 16.04.
Pixhawk USB Connection worked using Mission Planner 1.3.41.
However, it doesn't work after 1.3.42.
The system alerts the following error message.

ERROR MissionPlanner.MAVLinkInterface - System.NotImplementedException: The method or operation is not implemented.
  at MissionPlanner.MAVLinkInterface.getWPsfromstream (MAVLinkMessage& buffer, Byte sysid, Byte compid) <0x40372000 + 0x07ce7> in <filename unknown>:0 
  at MissionPlanner.MAVLinkInterface.readPacket () <0x40362000 + 0x037cb> in <filename unknown>:0 
  at MissionPlanner.MAVLinkInterface.getHeartBeat () <0x4041bdb0 + 0x0009f> in <filename unknown>:0 
  at MissionPlanner.MAVLinkInterface.OpenBg (System.Object PRsender, Boolean getparams, MissionPlanner.Controls.ProgressWorkerEventArgs progressWorkerEventArgs) <0x40412af0 + 0x00c6b> in <filename unknown>:0

I think "Marshal.ReadInt32(value, offset.ToInt32())" in getWPsfromstream function is cause this.
https://github.com/ArduPilot/MissionPlanner/blob/master/Mavlink/MAVLinkInterface.cs#L3958

Marshal.ReadInt32 is defined here.
https://github.com/mono/mono/blob/master/mcs/class/corlib/System.Runtime.InteropServices/Marshal.cs#L918

And, I found that the modification of this code is following one.
https://github.com/ArduPilot/MissionPlanner/commit/eea9f4f3b4b762257166542961ca26ac155c3783

I confirmed that Pixhawk USB Connection worked when I revert only the part to one's old way using MP 1.3.43.4.
Could you please fix this issue?

meee1, 2017-01-21:
thanks for do the full analysis on this one !

tatsuy, 2017-01-22:
Thank you!
I confirmed that it works well.

tatsuy, 2017-01-22:
Leave a comment

closed 2017-01-22 by tatsuy