ArduPilot/MissionPlanner #1462: Copter: Some sort of issue with Home not being set?

R-Lefebvre, 2017-01-22:
Trying to fly a Heli on AC3.4.1rc1.  Several attempts to plan a mission failed.  I would go to the mission planning screen, and the map would be fully zoomed-out and centered on 0,0.  A message pops up "Home Alt must be set" or something like that.  

I would then start building a mission, and it seems normal, until I add an RTL at the end, at which point the path lines disappear from the map.  If I try to write the mission to the vehicle, it fails and an unhelpful message like "Something went wrong" pops up.  Tried rebooting vehicle, and computer/MP, no change.

Randy thinks it's something to do with "Home not being set".  I'm not sure what the root cause of that is, but MP seems to not be handling the problem well, or giving the user a helpful message.

meee1, 2017-01-22:
i hate to say it. but did you set the home alt/home? it is required. it always has been.

MP needs it to calculate internal statistics/etc.
things that depend on the home positon/alt
1. inject gps
2. the distance calcs in flightplanner and flightdata
3. survey mission planning
4. spline wp calcs
5. uploading a flight plan
etc

R-Lefebvre, 2017-01-22:
I've never had to do that before?

meee1, 2017-01-22:
my guess is your settings reset for some reason. normally its saved between runs.

rmackay9, 2017-01-22:
The error message that is displayed when the mission upload fails isnt very helpful though.  It says something like "internal error" or "there is some problem" but doesnt tell the user they need to set the home position.  The user is left tompoke around trying different things at random until they figure out its the home position that is missing.

meee1, 2017-02-23:
added a new dialog c836f7f2

closed 2017-02-23 by meee1