ArduPilot/MissionPlanner #1466: About flight mode...Am I Right?

trigger1996, 2017-01-31:
http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html?highlight=flight%20mode
This official paper says we should modify MP ourselves, but I've tried another way, that is,  we can modify the parameters in the original MP in this place, changing the selected flight mode ourselves.
Tested, seems OK, but am i right?
[image]
[image]
Sorry for Chinese windows, may get you confused.

trigger1996, 2017-01-31:
First time using this, I put in the image twice.. -_-||

meee1, 2017-01-31:
yes, putting in the number will work fine.
it just will not display the correct name on the hud.

trigger1996, 2017-01-31:
Thanks, I fogot about this.
I just thought that it were impossible to select it in MP.

meee1, 2017-02-23:
you can modify paramatermetadata.xml file, and add your own mode, that will be used inside missionplanner.

closed 2017-02-23 by meee1