ArduPilot/MissionPlanner #1469: MP cant find libvlc.dll

n1kitap, 2017-02-03:
Im trying to make stream from X-plane to MP

I tried to connect vlc according to video https://www.youtube.com/watch?v=VYmkgcOw4ao

But MP shows error where he state he cant find valid libvlc.dll
[image]

Error msg via command line in MP
Caught exception of type Win32Exception while loading libvlc.dll from C:\Program Files (x86)\VideoLAN\VLC: %1 is not a valid win32 application

I have got latest stable MP and Windows 10 x64 (also tried on MP 1.3.42)
Latest stable VLC is installed in Program Files (x86)

After that when You close MP You will receive dialog (even if there is no any streams):
[image]

When You click either yes or no the MP will be closed.

Currently I figured out how to stream: Im using OBS Studio to capture video from Xplane to file, then VLC reencode and stream this file to http (according to http://vps.oborne.me/gcs/Mjpeg%20Video%20Source%20with%20VLC.htm )

Best regard Nikita.

meee1, 2017-02-06:
what are you trying to do? something does not make sense.

n1kitap, 2017-02-07:
Im trying to get video stream from VLC, how You showed in Youtube video "Mission Planner with a VLC supported video stream"

meee1, 2017-02-07:
please install the correct versions of vlc for your OS
if you have a 64 bit os you need vlc 64bit. etc
you may need to remove the the older version

n1kitap, 2017-02-07:
Thanks, everything is working after installing of VLC 64 bit. Now I successfully stream via UDP from OBS Studio, but with delay around 1.5-2 seconds. I ll try to use another capturing\streaming software. You can close this issue.

meee1, 2017-02-07:
yeah, vlc seems to buffer a lot.
using gstreamer can reduce the latency a lot.

closed 2017-02-07 by meee1