ArduPilot/MissionPlanner #1470: MP: Identification of a copter and internal storage of copter data

patschn, 2017-02-03:
Is there a way in the sourcecode of MissionPlanner to identify and to store values for a particular copter, as I don't want to access the memory of the copter directly. 

For example: I want to set values for a function, that only affects a specific copter, so the values can be loaded automatically if the copter has been connected to the pc.

Thank you in advance!

magicrub, 2017-02-08:
I don't think I understand your request..

patschn, 2017-02-08:
For example:
Let's say I have implemented a function, that fills a table with specifications of a copter, like batteryvoltage, batterycapacity, maximum/average speed etc. after the copter has been connected to the pc.
But in order to achieve this, I need a way to identify this copter (with an identification string or similar).

Therefore a way is required (in the source of MissionPlanner) to access this identification string and read it. Also this read data (batteryvoltage,capacity and so on) should be stored somewhere. What I discovered is Settings.Instance()['CustomString'] = 'YourValue'.
But is there something like Settings.Instance... where I can store copter specific data?

Simple:
A function reads the identification string of a connected copter (GetConnectedCopterID() returns "Copter: 1234").
Lookup coptervalues in storage. (LookupValuesInStorage("Copter: 1234") returns: (3.74V, 350mAh,...)

Does this clear up my request?

meee1, 2017-02-08:
the parameter BRD_SERIAL_NUM could be used to identify the copter/plane/rover/sub
this is a param, so can be accessed like all other params.

my first thought would be a param file with the filename based on the serial number.

patschn, 2017-02-11:
I'll try that. That seems to be exactly that parameter I need. Thank you!

closed 2017-02-23 by meee1