ArduPilot/MissionPlanner #1476: Export flight path as KML

SuperSharpShooter, 2017-02-13:
Is it possible to add a feature to export the KML of a flight plan rather than just giving the option of exporting it as waypoints?

There's a button 'View KML' on the right which downloads a file network.kml, which then contains a reference to http://127.0.0.1:56781/wps.kml which contains the information I would like to download.

Is is possible to allow a direct download of wps.kml using a button here?

meee1, 2017-02-16:
your pc does download that file to display it. the reason i assume its not shown is a setting on your pc. i dont have enough info to really tell you more

SuperSharpShooter, 2017-02-16:
The only file which is downloaded directly when the 'view kml' link in MP is clicked, is network.kml, which references http://127.0.0.1:56781/location.kml and http://127.0.0.1:56781/wps.kml. This means it is only able to display the kml while mission planner is open and serving it up. I would like to use the wps.kml file to export into another application once the flight path has been created. Can there be an option to download the wps.kml file directly rather than having to manually paste http://127.0.0.1:56781/wps.kml into the browser while MP is running and displaying the waypoints of interest ?

I have attached network.kml. (renamed to txt so it can be attached)
[network.txt](https://github.com/ArduPilot/MissionPlanner/files/779743/network.txt)

GiovanniVe, 2017-09-01:
Hi SuperSharpShooter; were you able to figure out how to visualize a flight plan without having MP running (nor installed altogether, for that matter)?
I'd also like to have that possibility.
Thanks
gio

SuperSharpShooter, 2017-09-01:
Nope, no luck.

samnorris92069, 2018-05-20:
Just save the "Place" in google earth once you load it.   Right click - save place as - and save it as a kmz or kml.  I believe it will save the current entries.