ArduPilot/MissionPlanner #1477: A problems with uploading firmware to the pixhawk

katzz0, 2017-02-13:
I updated Mission Planner up to 1.3.44 version but it seems there is a problem with uploading firmware to my pixhawk. I tryed quadcopter to flash 3.4.5. It downloads the firmware shows that it uploads it then it asks to replug a usb cable and starts flashing but the bar doesn't reach the end it stops and shows an timeout error. I downgrade Mission Planner to 1.3.33 (I chosed 1.3.33 to be sure so I did flashing before without any problems) and it works.

meee1, 2017-02-16:
did you have any other serial devices plugged in? please try removing them, and trying again.

katzz0, 2017-02-16:
No, I had only pixhawk connected. It points on the problem as I use the same envionment but the version was 1.3.33 and it worked.

meee1, 2017-02-23:
can you please upload the Missionplanner.log file for the failing version?

you can find the log in 
c:\programdata\Mission planner\