ArduPilot/MissionPlanner #1478: Problem with calibrating gps

PastaBiceps, 2017-02-15:
Hi there!

For a school project I am building a drone, this drone has the arducopter 2.6.
The arducopter is configurable with the missionplanner.
However when following the wizard I get stuck at calibrating the compass/gps, when I have to rotate it around the axis to all the white blocks, it suddenly stops and I receive the following error: "Setting new offsets for compass #1 failed". This happens at a random time when preforming the calibration. I cannot complete the calibration.

Any ideas/suggestions?

meee1, 2017-02-16:
please try the latest beta 
mp.
help > update beta

grkmogn, 2017-02-21:
i updated lastest beta mp but axis still does not rotate  and i do not make a compass calibration