ArduPilot/MissionPlanner #1481: Security and redundancy

charleslinquist, 2017-02-20:
With all the talk of reliability, redundancy and security,  I propose -

When a heartbeat is not received after 'x' seconds, Mission Planner would try to reconnect using a different protocol (from a user-defined list).

Mission Planner sets a flag when the program is first booted, and loads all parameters when it CONNECTs, then clears the flag and stores the SYSID.  After that, if a re-connect is requested and the flag is cleared and the SYSID has not changed, MP would not download all parameters (again).  This goes with my first request and would prevent a complete reload of parameters after redundant communication had been established.

Another protocol - SSH, should be added to tcp, udp and udpcl.   The use of SSH adds security to the connection without going through a proxy or dealing with port forwarding.

edwinhayes, 2017-02-22:
I am also quite interested in this.  I guess it depends though, on whether the design intent is that you only communicate over a single link at a time (and fall back between them), or run multiple links simultaneously?

@meee1 - what is the canonical way for dealing with, and managing, redundant links on the ground side?  I note you can add a second connection type by right clicking on the top bar - if you do this, are outbound messages sent over both links?  What if you then want to disable one of the links?  Can Mission Planner consider the link quality (from sequence numbers) of the links independently?  And are inbound MAVlink messages interposed, so you get duplicate sequence ID numbers and everything is handled twice, or are duplicate messages ignored?

charleslinquist, 2017-02-22:
When you right click on the bar and select a connection method, it 
disconnects the original connection.

Currently, I use a Microchip PIC (since I'm a LOT better with them than 
with ATMEL).  The PIC checks for valid Mavlink packets (by checking for 
the 0xFE start, capturing the packet and calculating the checksum).   
Currently, I use 4G as the 'primary' connection type,  since I also run 
1280 X 960 / 30 fps video over the 4G link.   If the 4G connection 
fails, I quit getting messages over that link.  Within a few seconds, 
the PIC notices that it no longer is receiving data over that channel, 
and if it is getting data over the 915Mhz (RFD 900+) link, it starts 
routing those packets to the PixHawk.  Currently, I wait 8 seconds of no 
data before I do any switch to prevent it from "chattering" back and 
forth.  The PixHawk always sends data over both methods (4G and 915Mhz) 
simultaneously.
I use 4G as my primary method, because if I lose 4G, I lose live video 
(5.8GHz doesn't have the range I want).  I fly mostly autonomously and 
I'm currently working on flying 22 miles and over a 3,000 foot mountain 
range with a quadcopter.  It isn't easy.   I think I can beat the FAA 
VLOS rule by flying close to it with a REAL helicopter over the route.

I have flown LOTS of non VLOS missions in Mexico (some places there have 
good 4G service!).  It is great having perfect video, telemetry and 
control when your craft is behind a 3,000' mountain range and is 6 miles 
away!

The method I use is hardly optimal, but I'm working out the glitches as 
fast as I can.  Since I'm sending data over 4G to a server using a 
"reverse SSH tunnel", the data from the PixHawk to the server is 
encrypted, but (currently) the data from the server to my laptop is 
not.  Theoretically, anyone who logs into "ec2-server1.xxxxx.net" could 
control my craft.  Not good. I'm working on adding SSH to that link.  I 
think the best bet for now is to add a software "shim" to my computer so 
it receives data over a secure link (SSH) and passes the data as tcp 
(localhost) to Mission Planner.   Mission Planner wouldn't need to be 
modified, but another piece of  software would be needed in the base 
computer.

My end goal is to modify Mission Planner (or have Michael Oborne and 
Co.)such that it has a new protocol - SSH.
But that will come a bit later.  First, I'm going to stop the reload of 
all parameters whenever a new connection is made to an existing flight.

Switching back and forth quickly and seamlessly takes a fair amount of 
intelligence.  I don't have all the answers yet, so I'm willing to 
listen to  your ideas.

I'm far from a software genius, but I have made small, notable changes 
in PixHawk code.  And I am very close to the point where I can compile 
Mission Planner, so I'll be able to make my own changes there - in time.

Tell me what you are up to?

Charles Linquist
Campbell, CA.

edwinhayes, 2017-02-22:
@charleslinquist - Hi Charles.  I'm the technology manager at Aeronavics, we're a ['boutique', I guess] industrial multirotor manufacturer in New Zealand.  At moment, we run a fairly standard Pixhawk/ArduCopter based setup with UDP for telemetry, but in the future we'll probably be looking at aircraft which have more than one link over which telemetry could be sent.  So I'm interested to know what people are doing, and whether the dev's have a picture themselves of how that type of functionality would be supported, so that I figure out how we should approach it.  Interested to hear what you're doing in terms of video transmission.  Unfortunately, 4G coverage here in New Zealand isn't that ubiquitous yet, and least not out in the countryside!

charleslinquist, 2017-02-22:
We have very good 4G coverage over most of the country.  Only in the
mountains does coverage get spotty, and that is why I need a radio at all.

I checked out your website.  Nice stuff!

I'm working on several projects right now, including a gasoline-powered hex
with 42" variable-pitch props and 200HP.  It is going together now.

charleslinquist, 2017-02-22:
You may be interested in my "data selector", although I'm redesigning it.
It currently looks for a valid MAVlink packet on two input ports.  It
favors port #1 (the PRIMARY). If a packet comes in (on either port), it is
buffered so its checksum can be calculated to determine if it is "good",
 it sets a flag and sends the data out to the PixHawk  If there is no valid
data coming in port 1, and there IS valid data coming in port #2, that data
is sent to the PIxHawk.  The PixHawk output data is always sent out using
both methods (and is not routed through my data selector).  If the data
should stop coming in on port #2, and the "good" flag was set for port #1,
the data from port #1 would again be sent out to the PixHawk.

You can do the same thing on the ground.  The data selector looks for valid
packets (again, favoring one over the other).

I currently cannot use my selector on the ground because radio data is
coming in over USB, while the 4G data is coming in over WiFi (I use my
phone as a hotspot to give me a 4G connection to my laptop).

I'm reworking the data selector to make it smaller and neater (it is
hand-wired now).  But the device might work great for you - on both ends -
if you have two RS-232 (TTL level) connections that you use now.