ArduPilot/MissionPlanner #1482: Geotagging Images - Option for GPS2 doesn't work.

edwinhayes, 2017-02-22:
There is a check-box for 'Use GPS2' in the geo-tagging form - it doesn't actually do anything (when using CAM/TRIG messages for tagging, the only information available anyway is the fused horizontal position).  Not sure what the purpose of this box is, but could probably be removed?

Also, what does the check box 'Use CAM messages' do?  You already select between CAM/TRIG and time offset on the line immediately above.

What there probably does need to be though, is a radio box to select tagging using either the fused altitude value in the CAM message, or the unfused GPS altitude value which is also included: the default should probably be fused, but for RTK applications you'd use the GPS instead.

edwinhayes, 2017-02-22:
Oh, v1.3.39.3 BTW.

meee1, 2017-02-23:
could you provide your dataset?

last time i tried it it was ok. hence the data set request

edwinhayes, 2017-02-23:
What do you mean 'it was ok'?  Since the CAM/TRIG messages currently don't have GPS2 information encoded (per my other thread), what would the expected behavior of the check-box be?