ArduPilot/MissionPlanner #1483: Rename the 'temp' form.

edwinhayes, 2017-02-22:
The ctl+f 'temp' window is clearly not that temporary - if contains things like georeferencing imagery and GPS injection which are pretty standard functions for plenty of use cases.

Can we rename this to something tidier, like "Advanced Functions", and provide some alternative method of getting to it than an totally non-obvious keyboard combination?

I think we already did this as a side effect of one of our theming related pull requests, but @meee1 cut it when merging, so I assume there was some objection?

meee1, 2017-02-23:
it is still temp, its just feature need to be moved out. for the general population.

meee1, 2017-02-23:
what feature specifically where you using?
georef is probly the biggest one in there that gets use afaik

meee1, 2017-02-23:
to put some stats behind it
temp accounts for 0.35% of all screenviews in MP
georef accounts for 0.10% of all screenviews in MP
these are from the last 30 days.

edwinhayes, 2017-02-23:
Interesting.  What counts as a 'screenview', or causes a screenview to increment?

But yes, I guess the title should be 'either rename the temp form, or move non-temporary functions somewhere else'.

The functions we're mostly using would be geo-tagging images (maybe a button on the log analysis tab in the flight data screen would be sensible?), and GPS injection (maybe on the 'actions' tab?).  Note that 'mostly using' actually means 'at the moment, number of our aircraft are being aimed at customers in the surveying industry, and these are the features they would be using'.