ArduPilot/MissionPlanner #1486: Scripting interface issue

DarkmatterVale, 2017-02-28:
Hi,

I am having a lot of trouble getting the scripting interface to work with the standard python modules installed on the machine. Is this a known issue with an easy fix? Below I have included the link to the specific conversation about the bug.

https://github.com/auvsi-suas/interop/issues/209

Please let me know if there is any other information that I need to provide.

meee1, 2017-03-10:
MP implement ironpython, and you can import external librarys
check http://stackoverflow.com/questions/1371994/importing-external-module-in-ironpython

DarkmatterVale, 2017-03-14:
@meee1 thank you for the assistance and I apologize for the delayed response.

It appears this has been fixed on the newest version of mission planner.

closed 2017-03-14 by DarkmatterVale