ArduPilot/MissionPlanner #1489: Request: Enhancement to Survey Grid - flight path is based on highest ground elevation for each path

ariffin22, 2017-03-03:
Hi developers,
   I'm not sure whether this feature has been requested before. I would like to request enhanced feature to survey grid. 

- Flight path/grid altitude is based on highest ground elevation
- This will maintain the average height between plane altitude and ground elevation hence improve average GSD.
- Plane will either descend/ascend before going to next path depending the highest ground elevation on the next path. please see example.

Appreciate if this could be implemented as it will improve quality of mapping data.
Thank you in advance.

example are as photos attached.

[image]
[image]

magicrub, 2017-03-05:
This can be added to a the GCS and then the mission be absolute altitude... Or.. you can use the implemented terrain feature and use the already implemented relative altitude.

Have you tried that yet?

jmachuca77, 2017-03-07:
Hi, can you tell what system those graphs below come from?

> Hi developers,
> I'm not sure whether this feature has been requested before. I would like to request enhanced feature to survey grid.
>
> Flight path/grid altitude is based on highest ground elevation [...]

Best regards,

Jaime Machuca Mercado
CTO | Droidika | www.Droidika.com
Mail. Jaime@Droidika.com
Tel. +52 (33) 3777 5327
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Droidika S.A. de C.V., deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

ariffin22, 2017-03-07:
@magicrub : 
   hi, yes i've tested it. the resulting flight path is as photo below as per your suggestion. Sometime with fix-wing, we can't get to target waypoint altitude in time. We won't have good overlap for stitching. It also can be dangerous if we fly low for higher GSD, if we can't get to intended altitude in time, the plane might hit the terrain. from my experience, if we can loiter to intended waypoint altitude, we'll have safer flight operation. so far i've done this manually for small area, adding loiter to altitude. but bigger area, it takes a lot of time. Sometimes i missed a few waypoints. 

[image]

@jmachuca77 :
   this is from Sensefly eBee simulator.