ArduPilot/MissionPlanner #1490: Sensor/Lidar/Rangefinder vs Baro only shows baro

OlivierJB, 2017-03-03:
In "pre-made" graphs, drop down menu

meee1, 2017-03-04:
did the log contain RFND messages?
the dropdown trys to graph 
BARO.Alt RFND.Dist1 RFND.Dist2

OlivierJB, 2017-03-04:
Just checked, no. I guess I am confused. Thought the dropdown would compare Baro alt and Salt from CTUN?
Here are 2 displays from same log . MP 1.3.44.3
Also http://discuss.ardupilot.org/t/sonar-rangefinder-check/15366  ?

[image]
[image]

meee1, 2017-03-04:
ill add salt to the message list

OlivierJB, 2017-03-06:
Sounds great, thanks!

closed 2017-03-06 by OlivierJB