ArduPilot/MissionPlanner #1496: Request: implement the option to choose between SRTM and Google Earth.

mfr84, 2017-03-10:
In many areas Mission Planner returns ground heights (SRTM) that deviate (incorrectly) from the values that you can see in Google Earth (usually more updated and more corrected).

So, it would be possible to implement a function that allows you to choose whether to use the heights derived from SRTM or those ones derived from Google Earth? 

It also would be possible to implement a function that allows you to choose whether to use SRTM data or Google Earth data when "Verify Height" is enabled ?

Thanks.

meee1, 2017-03-21:
the issue with google earth is there is a daily request limit. this is why it is not used.

mfr84, 2017-03-22:
Maybe the solution would be to allow the user also to pre-load its own DTM (available locally on PC) in order to be able to plan the flight based on that instead of retrieved online maps ? (A DTM can be easily extracted from Google Earth or other local SIT web portals...)

What do you think about it ?

meee1, 2017-03-22:
if you provide a geotiff, etc, this is already used instead of srtm.

mfr84, 2017-03-22:
Great. 

How I could import my DTM ? I cant find this option...

kikislater, 2017-05-06:
It's here : 
https://github.com/ArduPilot/MissionPlanner/issues/985