ArduPilot/MissionPlanner #1498: Request: Terrain-Following Management in Guided Mode

cbouju, 2017-03-13:
As stated in this post: http://discuss.ardupilot.org/t/terrain-following-works-in-auto-but-not-in-guided/15654/4 , "from an ArduPilot perspective, Guided does allow Terrain-Following"

Mission Planner manages the Terrain-Following for missions, but not for Guided Mode.

Thanks,

paulbroadmission, 2017-08-03:
Dear all,
May i know this has been solved or not? Tks a lot.

babakmtd, 2019-06-02:
in new version 3.6.8 that I Test,there is this problem and not solved.
terrain -following work in auto ,Poshold and AltHold but does not works in Guided mode.
my project is sprayer drone and use Rangefinder TF02 .
Randy,Please solved this bug in new version.

rmackay9, 2019-06-02:
This is really a flight code enhancement request (certainly not a bug) so I think I'll close this issue in the mission planner issues list.  I've opened [an equivalent issue in the flight code issue list](https://github.com/ArduPilot/ardupilot/issues/11467) including some advice to any developer who wants to pick this up.  It's not too difficult to implement but at the moment I really need to focus on object avoidance and non-gps navigation for Copter and Rover so I can't make any promises of when I will get to this.

closed 2019-06-02 by rmackay9