ArduPilot/MissionPlanner #1499: How do I add a custom flight mode? (Instructions say to raise an

AhmedSamara, 2017-03-15:
#### Issue details
http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html

Basically what I'm trying to make it so that I can press a button in mission planner, and then have my drone complete a simple pre-programmed aciton (ie: flying in a certain shape).

I thought this would be a straightforward thing to add it and make it appear in mission planner, but the instruction say to raise an issue here (apparently expecting someone to do it for me?).

Is there a straightforward way I can do what I'm trying to do without having to modify mission planner?

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2017-03-21:
you need to modify c:\programdata\missionplanner\paramatermetadata.xml

in here you will find the fltmode options. just follow the same structure and add as needed.

robemorin, 2017-03-27:
Hi.
I try to modify ParameterMetaData.xml, but when I ran Mission Planner some alert tells "Failed to get rally point" and MP doesn't show the new flight mode.

This is the part of code that I add.
   
&lt;FLTMODE4&gt;
&nbsp;&nbsp;&nbsp;&nbsp;<a style="color:#00FF00">&lt;DisplayName&gt;</a>Flight Mode 4&lt;/DisplayName>
&nbsp;&nbsp;&nbsp;&nbsp;&lt;Description>Flight mode when Channel 5 pwm is &gt;1490, &lt;= 1620&lt;/Description>
      &nbsp;&nbsp;&nbsp;&nbsp;&lt;Values>0:Stabilize,1:Acro,2:AltHold,3:Auto,4:Guided,5:Loiter,6:RTL,7:Circle,9:Land,11:Drift,13:Sport,14:Flip,15:AutoTune,16:PosHold,21:_My_New_Flight_Mode_&lt;/Values>
      &nbsp;&nbsp;&nbsp;&nbsp;&lt;User>Standard&lt;/User>
    &lt;/FLTMODE4>

lialiali, 2017-03-27:
@robemorin 
hello, I also want to do a similiar thing.Please tell me which software and code should I need to install .
and I can't find the"c:\programdata\missionplanner\paramatermetadata.xml",

magicrub, 2017-03-27:
What flight mode are you wanting to add?

And btw, adding it on the GCS is basically just adding the text of the name. The actual flight mode code must be in the flight controller.

lialiali, 2017-03-27:
I want to add a mode just like if I push the button on the handle,it could fly a round or so.
and which software and code should I need?
thanks

lialiali, 2017-03-27:
Now,I use the visual studio to see the source code mission planner,can I use visual studio to add the mode?
And do you now which file is about the flight mode control?
thanks a lot!

magicrub, 2017-03-27:
.. that's called auto.

lialiali, 2017-03-27:
yes,do you know how could I achieve this

robemorin, 2017-03-27:
Hi @lialiali and  @magicrub .

I found the file in `c:programsfiles(x86)/MissionPlanner/ParameterMetaData.xml`, you can modify the file with notepad or notepad++.

I want to add a one more flight mode, I cloned STABILIZED as MySTABILIZED (I want  to modify later), and I was following the instructions in [Ardupilot](http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html), but I can't do it the last step.

rmackay9, 2017-03-28:
Maybe we should modify that final suggestion in the wiki.  If the flight mode is only for your personal use then modifying the .xml file is the way to go.  If you want it to be a flight mode used by everyone then really what needs to happen is the flight mode needs to be submitted as a PR to ardupilot.  Once it's in and released, the new flight mode will appear automatically in the MP's flight mode drop-down lists (and on the HUD).