ArduPilot/MissionPlanner #1500: Compass Calibration in V 1.3.44 build 1.1.6240.11550

SpoonFPV, 2017-03-16:
#### Issue details
The compass calibration setting offsets fails with the current build of 1.3.44

#### Version
V 1.3.44 
Windows 7 x64

#### Platform
[  ]Pixhawk
[  ] Copter

#### Airframe type
 H4

#### Hardware type
Pixhawk with external Ublox compass.  Failure is consistent on 5 vehicles, ran the calibration on version 1.3.38 and the calibration worked fine.  Very confident it is a software issue.

#### Logs
No Logs available.

meee1, 2017-03-16:
onboard or external? and what error do you actually get?

SpoonFPV, 2017-03-16:
Both internal and external I get "setting new offsets for compass #1
filled,  and same message for #2, not running a 3rd.  Works in other
versions of software, failure was across 5 different platforms in two
different outdoor locations.

meee1, 2017-03-16:
what firmware version?