ArduPilot/MissionPlanner #1501: MissionPlanner 1.3.45 cannot connect to PixRacer ESP8266

Frontier314, 2017-03-20:
#### Issue details
I have an AUAV.co PixRacer running ArduCopter 3.4.5. The PixRacer has an ESP8266 WiFi board which is running MavLink ESP8266 firmware v1.1.0. I am using Mission Planner 1.3.44 on two machines (one notebook and one Intel tablet), both running Windows 10 Pro x64 and using Intel PROwireless driver (the latest version from Intel). I can connect without issues to the "PixRacer" AP via Mission Planner 1.3.44. The connection settings are UDP at 921k baud, using the standard connection ports.

Upon upgrading to Mission Planner 1.3.45 (tried both in-place upgrade and uninstalling/re-installing the software), I've noticed a change to the connection behaviour; Mission Planner upon finishing loading tries immediately to connect via UDP to the PixRacer, even if the board is not initialized. Subsequent efforts to connect to the board (after powering it up) fail and it is no longer possible to connect to the PixRacer via WiFi. Connecting via USB or 3DR MavLink telemetry works as expected, only the UDP WiFi connection is no longer possible. Shutting down Mission Planner, rebooting the machine, rebooting the PixRacer, connecting the PC to the PixRacer AP and then starting up Mission Planner does not change anything; I cannot connect to the PixRacer via WiFi.

I have reverted back to MP v1.3.44 where everything works fine but I would like to know if it is possible to mimic the v1.3.44 connection behaviour on MP v1.3.45 in order to be able to connect via WiFi to the PixRacer.

#### Version
1.3.45

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad X

#### Hardware type
AUAV PixRacer

Thank you very much in advance for your help.

meee1, 2017-03-21:
MP will try autoconnect on startup.
if it fails at first you can manual connect afterwards and it should work just fine.

Frontier314, 2017-04-08:
Sorry for not replying earlier, I was away due to work.
Unfortunately, I cannot re-connect to PixRacer with 1.3.45.

No matter if the Windows PC (tried 4 different ones with Windows 7 x64 and Windows 10 x64 OS) is connected to PixRacer WiFi prior launch or not, it is impossible to connect via UDP connection. 10 times out of 10 MP after asking for the UDP port, it simply starts the device connection timeout requester (30 seconds) and never connects.

Closing MP, uninstalled 1.3.45, reinstalled 1.3.44, launched MP, CONNECT via UDP (the PC is connected to PixRacer WiFi the whole time) and it connects immediately. Between version up/down-grades, the application is fully uninstalled and both program and program data folder is removed.

I am attaching Mission Planner logs from both versions (1.3.44 and 1.3.45), v1.3.44 connects every time where 1.3.45 does not connect at all. I also see that Randy Mackay has opened a similar issue #1511, maybe these two issues are related.

Again, thank you very much for your help.

[MissionPlanner_1344.txt](https://github.com/ArduPilot/MissionPlanner/files/907760/MissionPlanner_1344.txt)
[MissionPlanner_1345.txt](https://github.com/ArduPilot/MissionPlanner/files/907761/MissionPlanner_1345.txt)

Daddy-Boomalati, 2017-10-03:
I am also using a PixRacer and cannot connect via UDP. I am using the latest versions of Copter and MP. When I try to connect I use the default listening port (14550) and click "connect", but it just says "Connecting to MavLink" and nothing happens.

meee1, 2017-10-03:
do you have more than one network card I your pc? if you have more than one this can cause issues as to which interface MP listerns on

Frontier314, 2017-10-04:
@meee1 

I am using 4 different Windows PCs with MP (all of them run Windows 10 build 1703, three of them are using the x86_64 variant and one is x86). The three PCs are laptops (x86_64) so they have at least a wired Ethernet card and a WiFi card. The other one (x86) is a tablet which has only a WiFi network card.

On all four of them, I can reproduce the problem every single time: install MP 1.3.49 (or 1.3.45) and it cannot connect to the PixRacer via ESP8266 (the "PixRacer" access point). Remove MP 1.3.49, install MP 1.3.44 and it connects immediately (the same machine).

The logs attached above are from that simple test.

If you want me to perform some other test, let me know.

meee1, 2017-10-09:
this appears to be be a esp8266 firmware version issue. (forum discussion)

possible because of the compiled in mavlink headers limiting its function.

closed 2017-10-09 by meee1