ArduPilot/MissionPlanner #1504: RTK/GPS Inject screen mockup

rmackay9, 2017-03-22:
Here's a mockup of what the RTK/GPS Inject screen might look like.  I think it's easier for people to understand but I leave it to your better judgement of course.

[image]

proficnc, 2017-03-22:
Nice Randy!

If we could also add a force survey in button, that would be great.

ACPUK, 2017-04-21:
Looks good Randy, however is there still going to be a GPS, Glonass and Beidou indication like there is now. 
Also the bar graphs of the sat strength / quality. The bar graphs could be much smaller then they are now.
Is is nice to see the info there.

Kiwa21, 2017-04-26:
I think saving and displaying the accuracy of the saved data is a real plus to help user using recorded position.

Naterater, 2019-01-29:
I think this still needs some work. I've never done the "survey in," however I almost always connect to NTRIP corrections. With the HerePro and less expensive L1/L2 receivers coming, more and more users are going to be using this. 

1. Add Galileo to the RTCM list and give an indication if it is receiving or not. 
2. Add different colors or additional bars or something for differentiating between L1, L2, and L5 signals. These are all coming.
3. Increase the Messages Seen box size. My NTRIP provider sends them all.
4. Improve the GUI for NTRIP connection information, and poll for the available mounts. Many other programs give a list after entering the address and port.
5. There is an error box that shows every time a connection is initiated, but then the next window is canceled. Canceling a connection isn't really an error. 
[image]