ArduPilot/MissionPlanner #1505: Allow Video and HUD to be seen at the same time

rmackay9, 2017-03-23:
With the latest [APSync](http://ardupilot.org/dev/docs/apsync-intro.html) testing, the video from the camera appears immediately in place of the HUD.  This is very convenient but the HUD has a lot of important information on it including the EKF and Vibration monitors.  Also it shows the current flight mode and any pre-arm checks.

It would be nice if the Video and HUD could be displayed at the same time. Maybe we need a new window?

suraqis, 2017-08-04:
I believe we need right panel with Video and Tab panel. Also it will solve issue with quickveiw indicators are not visible when tabs are switched to Actions or other. But it works well on Full HD displays. If you have 1366x768 it will be very narrow map area. But un wide display it looks good. Something like [this ](http://imgur.com/a/cM20A)

datasetsk, 2021-06-04:
Any idea how to edit using C# by adding an additional panel to the left along with hud and quick action tabs