ArduPilot/MissionPlanner #1508: If I can add a new fly model in Visual studio and I have had an source code of MP

lialiali, 2017-03-27:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-03-27:
Have you tried using the softwares as is? It does all of that. Please read the wiki.

lialiali, 2017-03-27:
@magicrub 
Thank you.But I want to add a new model to achieve a different function, like to make it to control itself to the trajectory flight .not to clickon the map in software. So, how could I add this model?

magicrub, 2017-03-27:
mode AUTO will fly a pre-programmed mission of pretty much anything you want
mode GUIDED will fly to a point than an external system is feeding it trajectories/vectors/locations

and again, this is all in the autopilot, not the ground station software, so i suggest you close this issue (and the other duplicate issue) and post help question on discuss.ardupilot.org