ArduPilot/MissionPlanner #1509: MP 1.3.45 SITL cannot connect to X-Plane & anti-virus issue.

SwiftGust, 2017-03-30:
#### Issue details
[image]
I'm trying MP SITL with X-Plane, when i try to run pre-built firmware, 
after MP says Downloading sitl software
gives me this error, sorry about the korean(it says access denied).
I can't get SITL working due to Firmware binary is recognized as virus to my panda antivirus.
(Solved this with exclusion tho...)

**and main problem is that MP tries to connect TCP with X-Plane, so It cannot connect.**

P.S should I necessarily point home in SITL? received data shows correct GPS Location in X-Plane.

#### Version
MP 1.3.45 build 1.1.6286.16738

#### Platform
[V] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

jmachuca77, 2017-03-30:
Did you follow these instructions to get SITL working:

http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html <http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html>

Best regards,

Jaime Machuca Mercado
CTO | Droidika | www.Droidika.com
Mail. Jaime@Droidika.com
Tel. +52 (33) 3777 5327
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Droidika S.A. de C.V., deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

SwiftGust, 2017-03-30:
http://ardupilot.org/dev/docs/sitl-with-xplane.html
I followed this document and X-Plane specifically only work with UDP port... Not TCP...
and current MP-SITL automatically sets connection only to TCP.