ArduPilot/MissionPlanner #1511: UDP re-connection unreliable

rmackay9, 2017-03-31:
Since I've started using APSync's wifi telemetry I've found the MP's UDP connection can be unreliable.

In particular this sequence makes it impossible to reconnect via UDP until MP is restarted:

- connect PC via Wifi to APSync's wifi access point (aka "ardupilot")
- on MP, select UDP and Connect (connects successfully)
- on MP, disconnect
- on MP, connect.  A message box, "Connect Failed" appears immediately
[image]

Daddy-Boomalati, 2017-10-03:
I am using a PixRacer and cannot connect via UDP. I am using the latest versions of Copter and MP. When I try to connect I use the default listening port (14550) and click "connect", but it just says "Connecting to MavLink" and nothing happens.

meee1, 2017-10-09:
this appears to be a esp8266 firmware issue with regards to the pixracer.

i believe the original issue is also resolved so long as the disconnect button is used. as you can only listen on port 14550 once.

closed 2017-10-09 by meee1