ArduPilot/MissionPlanner #1514: Missing Heartbeats with UDP connection

kas219, 2017-04-03:
When connecting to Pixhawk using a UDP connection, the heartbeat messages are only sporadically sent to the autopilot. This results in the autopilot believing there is a loss of link and in our case performing an RTL. The connection to the aircraft is still rock solid as there are few if any dropped packets and the autopilot can still be commanded without delay. I have verified the packets are not being sent by monitoring the connection with Wireshark.

I have verified this problem dating back to at least version 1.3.39. I have not tested versions prior to this.

meee1, 2017-04-10:
could you provide a tlog and wireshark capture? of the same session? 
the tlog should show the outgoing hb's where as the wireshark should show it being transported beyond that

meee1, 2017-04-10:
I cant see any issues
[image]

kas219, 2017-04-11:
I have a Wireshark and TLOG captured that show the problem but I am not being allowed to attach the files. I have put them in a zip file but it will not let them be attached.

Keith