ArduPilot/MissionPlanner #1515: MP error on launch

chris669, 2017-04-05:
#### Issue details
Hi, three weeks that I try to launch MP with no success. I desinstalled it and reinstalled. See the error message on the attached pict. Error that appears after some minutes...
I try other software like APM and QGC and I have not this problem. I forwared the error messages when I was invited to do that, but no answer at this time. It's possible that I desactivated some required by MP Windows services because I try to optimized my Windows.

#### Version
1.3.45

#### Platform
Windows 10 64bits
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
X4_

#### Hardware type
pixhawk

#### Logs
see pict

[image]

meee1, 2017-04-10:
does MP continue to run after this error?

chris669, 2017-04-12:
Ok, I have to wait around 4 minutes in order to be able to use MP. If I wait without doing nothing, I do not have the error displayed in the above printscreen. I was invited after the 4 minutes to download the latest version. Now running 1.3.4.6. It seems your app is looking for something that it couldn't find on my PC. A missing service? How can I help?

meee1, 2017-04-12:
do you use windows 10 with a intel onboard graphics card?

chris669, 2017-04-12:
Yes, Graphique HD Intel, on a HP

hackra76, 2017-04-12:
One of the latest windows 10 updates breaks the Intel graphics driver, you have to reinstall it... It worked for me, I used the Intel driver update utility from Intel website.

chris669, 2017-04-13:
Thanks a lot hackra76, that was the solution for me.

closed 2017-04-13 by chris669