ArduPilot/MissionPlanner #1517: Bug: Survey tool Invalid Entry Error

iskess, 2017-04-10:
#### Issue details
When pressing Accept in the Survey Tool, an error window appears that says, "A invalid entry has been detected, Invadid number on row 1".
There is on OK button but nothing happens when it is pressed. I can not dismiss the window. I had to use CTRL+ALT+DEL to get past the error. 

#### Version
MP 1.3.46

[image]

iskess, 2017-04-10:
If I click OK enough times (one for each WP I presume), the dialog will disappear, but no flight lines will show. Downgrading to 1.3.45 resolves the problem

meee1, 2017-04-10:
ok ive resolved this. and am pushing a replacement

closed 2017-04-15 by meee1