ArduPilot/MissionPlanner #1518: Gimbal on Pixhawk

Phil1988, 2017-04-11:
#### Issue details
Yesterday I wanted to setup my 3 axis [Strom32](http://www.firstquadcopter.com/gallery/hakrc-storm32-images/Storm32-BGC-v1.3-pinout.jpg) gimbal with Mission Planner for my Pixhawk.
The goal was to link it via Mavlink on Serial 4/5 to the UART of the Pixhawk.
Because I got 0 response from my gimbal I started to go a easier way:
I took the AUX channels to send PWM signals to RC2-0 (pitch/tilt I call it TiltInput) and RC2-2 (pan/yaw I call it PanInput).

I went to Initial Setup->Optional Hardware->Camera Gimbal
And everything was disabled.
I set the Type to "Servo".
checked stabilize Pan and stabilie Tilt.
Set Tilt to RC9 with input RC7.
Set Pan to RC10 with input RC8.
Took the Radio Controller and moved RC7 => Tilt AND Pan moving.
Took the Radio Controller and moved RC8 => Nothing happens.

I frickled around and could do the following:
Disable all.
Click on Config and then back to  Initial Setup->Optional Hardware->Camera Gimbal (so that the dropdowns are empty).
Enable Pan with output RC9-RC11 (connected with PanInput of Strom32) and input RC8 => only Pan is moving.
Enable Pan with output RC9-RC11 (connected with PanInput of Strom32) and input RC7 => only Pan is moving.

Disable all.
Click on Config and then back to  Initial Setup->Optional Hardware->Camera Gimbal (so that the dropdowns are empty).
Enable Tilt with output RC9-RC11 (connected with TiltInput of Strom32) and input RC8 => only Tilt is moving.
Enable Tilt with output RC9-RC11 (connected with TiltInput of Strom32) and input RC7 => only Tilt is moving.

Disable all.
Click on Config and then back to  Initial Setup->Optional Hardware->Camera Gimbal (so that the dropdowns are empty).
Enable Pan with output RC10 (connected with PanInput of Strom32) and input RC8 => only Pan is moving.

NOW: 
Tilt is still disabled (empty drop downs for its Output and Input).
Set the Output of Tilt to RC9 and Input to RC7 => Pan AND Tilt are moving on using Channel 7!!
If I use channel8 of my Remote Controller then nothing happens.

It looks like there is something messed up.
There must be a way to set this right.

So all I want to do is set Channel 7 to Pan (or Tilt) and set Channel 8 to Tilt (or Pan).

If we figured out how to set this up correctly, then the next step would be to get my initial goal: 
Use a mavlink connection via Serial4 to the Strom32.

I hope you can help me, because there must be an issue with Mission Planner and the Doc about Gimbal with Servos (http://ardupilot.org/copter/docs/common-camera-gimbal.html) is missing the Pan part.

Also this (http://ardupilot.org/copter/docs/common-storm32-gimbal.html) is not working with TELEM2 OR Serial4....

#### Version
V3.4.6

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Hex

#### Hardware type
Pixhawk

lvale, 2017-04-15:
I would get this request for help on the ArduPilot forum. Also be sure to include which settings you have your gimbal configured, including gimbal firmware

closed 2017-04-27 by meee1

Zabbie1, 2018-04-13:
have you solved your issue yet?

Phil1988, 2018-04-13:
Not via AUX...
I somehow managed to get MAVLINK working and had no issues with this problem.

But I stombled over so many issues during this process that I really have no idea what will come next to you :D

Zabbie1, 2018-04-14:
i can help you with controlling the gimbal pan tilt roll using the AUX of pixhawk if you want.

sharoseali, 2018-11-21:
@Zabbie1 , your help is required he

agising, 2019-04-12:
Me too, did you solve it?