ArduPilot/MissionPlanner #1519: Gstreamer Setup File Link generates uncaught exception

Loran425, 2017-04-11:
#### Issue details
Mission Planner Detects a stream and brings up the new install/config prompt the link on this tab is broken

#### Version
1.3.46
build 1.1.6310.11669

#### Platform
- [x] All 
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover
- [ ] Sub

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
[image]

[Gstreamer_install_link_error_log.txt](https://github.com/ArduPilot/MissionPlanner/files/914611/Gstreamer_install_link_error_log.txt)

meee1, 2017-04-11:
this error normally indicates an issue on your pc. with the default browser association.
the link is this
https://gstreamer.freedesktop.org/data/pkg/windows/1.9.2/gstreamer-1.0-x86-1.9.2.msi

Loran425, 2017-04-11:
Ahhh that sounds about right. I found the file just fine. Just wanted to make sure the error was noticed since that dialog is new with the release today.

closed 2017-04-11 by Loran425