ArduPilot/MissionPlanner #1525: Channel value as RSSI

Phil1988, 2017-04-20:
#### Issue details
Its a funny thing, that its mentioned here: https://pixhawk.org/users/rc_rssi, that most vendors didnt include RSSI into their protocols but also Pixhawk, ArduPilot and Mission Planner dont have the option to take one channel as RSSI value.
I am building the SBUS signal myself and dont like to do additional PWM or D->A converting to get RSSI working.

Instead I would like to take channel 12 as RSSI value for my pixhawk.

I can set RSSI_ANA_PIN to 103 (to select SBUS) but this is meant to use the SBUS output of Pixhawk to read the PWM value there....  (really ? -.- again "Pixhawk SBUS" for RSSI takes the SBUS OUTPUT and wants a PWM ?!?! )
Would be great if there is an option to take a channel value and use it as RSSI.

#### Version
Mission Planner 1.3.46 build 1.1.6309.37214 APM: Copter V3.4.6

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Hex

#### Hardware type
Pixhawk

meee1, 2017-04-27:
what format is the rssi in? analog or pwm? MP supports displaying rssi in it is sent via mavlink. this question would be better asked on the forum I believe

closed 2017-04-27 by meee1

Phil1988, 2017-04-28:
It's not really a question because its just impossible to send the RSSI value to the FC as an channel (SBUS,PPM...)
As I wrote in my first message, I want to use a channel value to be the RSSI value - because normally there shouldnt be any needs to give the RSSI out of the Rx as a seperate signal (either analog 0-3.3V or PWM) but the existing wire (SBUS or PPM) could include the RSSI value.

So it should be possible to tell the copter to use (for example) the imput value of channel 12 (or any other) to be the RSSI value.