ArduPilot/MissionPlanner #1526: Can't open MP on latest win10.

armadyne, 2017-04-21:
Recently Microsoft pushed me the latest win10 update, after updating the system mission planner can not be used, it will crash when I start up, please help!

meee1, 2017-04-27:
can you please post more info on the issue.
logs/screenshots etc
does it hang? does it not start at all? do you have an intel graphics adapter.

armadyne, 2017-04-28:
Thanks for replying. I have already rolled back to the previous win10 version and MP is working again. My CPU is 4790K but I don't use integrated graphics card I use GTX970 instead. When using the latest win10 version it just crashed when I starts up.

rmackay9, 2022-11-30:
I think we can close this because MP seems to be working fine on Win10.

closed 2022-11-30 by rmackay9