ArduPilot/MissionPlanner #1528: MP - updater crashes

lvale, 2017-04-23:
Latest version of release MP on clean Windows 10

Trying to do the update by the Update to latest beta and Updater crashes

Reboot the machine and tried again, same issue

[image]

lvale, 2017-04-23:
```
<?xml version="1.0" encoding="utf-8"?>
<configuration>
<startup><supportedRuntime version="v2.0.50727" /></startup>
  <runtime>
    <assemblyBinding xmlns="urn:schemas-microsoft-com:asm.v1">
      <dependentAssembly>
        <assemblyIdentity name="System.Runtime" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-4.1.1.0" newVersion="4.1.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="System.Diagnostics.Tracing" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-4.1.1.0" newVersion="4.1.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="System.Reflection" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-4.1.1.0" newVersion="4.1.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="System.Runtime.InteropServices" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-4.1.0.0" newVersion="4.1.0.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="System.Diagnostics.DiagnosticSource" publicKeyToken="cc7b13ffcd2ddd51" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-4.0.1.0" newVersion="4.0.1.0" />
      </dependentAssembly>
    </assemblyBinding>
  </runtime>
</configuration>
```

[image]

rmackay9, 2017-04-24:
I've seen this as well actually.

hackra76, 2017-04-24:
Happens to me also, Windows 10 after latest Creator update ....

Rolf-G, 2017-04-25:
Happens to me also: win10

meee1, 2017-04-27:
this should now be resolved. I think

SSUAV, 2017-04-28:
Yep, Win 10. MP Crashing at startup
[image]

meee1, 2017-04-29:
@SSUAV different error than the original post
but partial related. are you running beta? 90% of these issues are beta issues

SSUAV, 2017-04-29:
Yes, sorry about that. Reinstalled MP working good!! Thanks!!

lvale, 2017-04-29:
sorted. closing. 
Thanks Michael

closed 2017-04-29 by lvale