ArduPilot/MissionPlanner #1530: Add battery cell voltages

rmackay9, 2017-04-26:
Copter-3.6 (still many months from release) will include additional battery information in the BATTERY_STATUS mavlink message (http://mavlink.org/messages/common#BATTERY_STATUS):

- temperature (in centi-degrees)
- individual cell voltages

The relevant PR is here: https://github.com/ArduPilot/ardupilot/pull/6106
The code that sends this BATTERY_STATUS message is in the "send_battery_status" method in the GCS_MAVLink library's GCS_Common.cpp file.  https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L237

meee1, 2017-04-26:
already in
https://github.com/ArduPilot/MissionPlanner/commit/7570ec0ee1dc0dfa16619b347b141c555fc7d3cb

rmackay9, 2017-04-26:
Great!

closed 2017-04-26 by rmackay9