ArduPilot/MissionPlanner #1531: PreFlight Checklist sorting

iskess, 2017-04-26:
#### Issue details
I really like the preflight checklist feature. It would be more usable if we could re-sort the list using the editor. 

I'm not clear what the + button does to the right of each line. The - button deletes the line, and the + button adds another line but without a Description/Value field. What is this used for? It seems to add an additional condition, but I cannot get it to work.  It does not insert another checklist item in between as I had hoped. 

#### Version
1.3.46

iskess, 2017-04-26:
Another suggestion: I would like to check a few parameters before each flight, like the RTL altitude, and whether Terrain Following is active, and which Failsafe mode I'm in. Is it possible to add the parameters to the list of variables to display their value?

iskess, 2017-04-27:
I tried to edit checklist.xml or checklistDefault.xml, but neither are being used by MP. Where is the PreFlight checklist file hiding?

meee1, 2017-04-27:
mydocuments\missionplanner\checklist.xml  < edit that one
as for the additional + that's for compound conditions

iskess, 2017-04-28:
Thanks. Any chance you can allow for inserting or changing the sequence?

jagrandle, 2017-04-28:
I don't even have this file    mydocuments\missionplanner\checklist.xml       !!!!
Thus I am not sure how I can change it :)
Thanks