ArduPilot/MissionPlanner #1533: Image Tagging and Here + RTK  using Baro Alt not GPS Alt

leondwyer, 2017-04-28:
#### Issue details
Currently, image tagging is using int32_t  altitude; (baro alt)
this makes sense using a regular GPS. However, with better positioning, it would be good to have a checkbox to select that would make MP tagging use the altitude_gps record instead.

#### Version
current beta vertion

#### Platform
Plane

#### Airframe type
Delta Wing

#### Hardware type
Ph2.1   Here +

#### Logs
NA

meee1, 2017-06-02:
what exact settings are you using?

geofrancis, 2017-09-30:
i have the same issue, i would like to use the gps altitude from the RTK for tagging not baro altitude.

aloerch, 2017-11-06:
This is ultimately my preference too. GPS Alt, based on an RTK, will should give better accuracy than either baro alt or AMSL based on EGM96.

meee1, 2017-11-06:
are you using cam messages or gps messages? I need more information

aloerch, 2017-11-06:
I'm using CAM messages.