ArduPilot/MissionPlanner #1537: Custom Parameter List Page.

iskess, 2017-05-01:
What would be really useful is a custom tuning page where we could organize any parameters we want on the page for quick reference and easy adjustment.  Although the find feature helps tremendously, it is still cumbersome to locate often changed parameters quickly, especially while flying.

robustini, 2017-05-19:
Absolutely VIP feature!
It would be very useful to select from the "full parameters list" by flag the list of parameters to be displayed with that filter, and also be able to enter a custom description of each, for example, to record comments pertaining to each single parameter during tuning or other.

Ironcenturion, 2018-01-16:
Agreed- I'm using MP and APM:Plane in a college training course, we have several parameters to verify prior to flight (ALT_HOLD_RTL, etc) and it would save time and potential mistakes to have a "custom" menu with just a handful of parameters for easy reference.