ArduPilot/MissionPlanner #1538: Can't get Advanced Params and Full Parameter List menu

weety, 2017-05-01:
#### Issue details
Can't get Advanced Params and Full Parameter List menu. I won't midified MOT_THST_HOVER and MOT_HOVER_LEARN parameters, but i can't do it.

#### Version
Since Mission Planner 1.3.42 version

#### Platform
[  ] All
[  ] AntennaTracker
[* ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
pixhawk, firmware version arducopter v3.4.4

#### Logs
_Please provide a link to any relevant logs that show the issue_

weety, 2017-05-01:
Airframe type is quadrotor.

iskess, 2017-05-01:
Try: Config/Tuning > Planner >Layout > Advanced

weety, 2017-05-02:
Thank you very much!

closed 2017-05-02 by weety